PrintersForAnts/Mini-AfterSherpa #14: Support for Orbiter V2 + Revo

failsafe-fpv, 2022-04-14:
Hey, this isn't an issue but I was just wondering if you've got a duct which supports an orbiter v2 and/or E3D Revo Voron hotend?

Cheers,

RC

KurioHonoo, 2022-04-14:
Oh god, I've neglected this repo severely. 

Depending on which Revo model you have the standard size one works with any of the V6 style MiniAS models since the Revo is a drop in V6 replacement. I have a version for the Micro that I'm currently using on my V0, but it's unreleased. As for the Orbiter V2, I'll have to double check since I've made a fully custom cowling and x carriage for someone once as an example and I know that I've worked on a mounting plate adapter as well. 

I have a tendency to make new versions and share them in my channel on DOOM or with clee and then I take forever to get it uploaded here.

KurioHonoo, 2022-04-14:
Reach out to me on Discord, I'm Kyrios on Voron and DOOM.

KurioHonoo, 2022-04-30:
Revo Voron support added with https://github.com/KurioHonoo/Mini-AfterSherpa/commit/22ea37053a96fb3ac591a052a6c5b94502d981a9

As for Orbiter V2 support.... oh hey, I think I hear my cat calling me.

failsafe-fpv, 2022-04-30:
Awesome, oh so there isn't an orbiter V2 mount available yet then?

KurioHonoo, 2022-05-07:
I currently have a the Dragon and a V6/Zodiac BMO version. Revo Voron will be an easy change though. I'm still not sure that I like so many different versions for this, but I guess it's really not that big of a deal.

jtrmal, 2022-05-07:
maybe that's how it is if you have so little space to work with :)

KurioHonoo, 2022-06-29:
Yeahhhhh sorry about the wait, but I did the thing. https://github.com/PrintersForAnts/Mini-AfterSherpa/commit/aa22476e9e41db300028ee4784a6ece3717b9866

closed 2022-06-29 by KurioHonoo