PrintersForAnts/Mini-AfterSherpa #15: Support for Revo Voron

adadrag, 2022-04-17:
Since the Revo voron cold mount is out on market it will be very nice to have a mount for it.

KurioHonoo, 2022-04-18:
If you can find me a CAD/step file or even an STL of it then I can get right on it because I haven't been able to find anything. Otherwise I'm just going to have to wait until I get mine in and make the model myself.

Deathzo, 2022-04-25:
Will this work? I tried my cad-fu but uh yeah I'm lvl 0.01 with cad. Watching Solidworks videos though!

[](https://github.com/VoronDesign/Voron-0/blob/Voron0.1/STLs/Toolheads/Mini_Afterburner/Revo_Voron_Toolhead_DD/%5Ba%5D_Cowling_Revo_Voron_MULTIBODY_x1.STL)

KurioHonoo, 2022-04-30:
Added! https://github.com/KurioHonoo/Mini-AfterSherpa/commit/22ea37053a96fb3ac591a052a6c5b94502d981a9

closed 2022-04-30 by KurioHonoo